OpenHD/QOpenHD #1: Video decoding/rendering slow

steveatinfincia, 2019-10-12:
The `master` branch is using QtGstreamer, which is old and unmaintained but generally works. However it uses `avdec_h264`, which is a software h264 decoder and hits the CPU hard, especially on low end devices. In the worst case I have seen it delay video by almost 30 seconds.

In the `qmlglsink` branch, I have added support for a more modern rendering system. It still uses `avdec_h264` at the moment and needs some simple changes to enable the hardware decoders for each platform. I haven't done so yet because I can't get the gstreamer androidmedia plugin to load for some reason, and `vtdec` (the Mac/iOS hardware decoder) shows some jitter.

steveatinfincia, 2019-10-20:
I've discovered the reason for the `androidmedia` plugin issues (and therefore the hardware decode acceleration issues on Android).

The `androidmedia` plugin is a complex bit of "glue code" that has to call out to Java via JNI in order to discover the available codecs on a particular Android device when the app first launches. In order to do that there is some build integration necessary, and Qt Creator/qmake ignores those steps at the moment.

I will have to take a look at what they're doing and add the appropriate build steps, but once that is done everything should work perfectly.

This is the only thing holding up integration of the more modern `qmlglsink` video rendering system, it works very well everywhere else.

Reference: https://lists.freedesktop.org/archives/gstreamer-android/2016-September/001138.html

pilotnbr1, 2019-10-20:
Very nice!

steveatinfincia, 2019-11-21:
Hardware decode on Android is working well now, but the gstreamer `decodebin` element that automates the process of discovering available codecs currently has some quirks that prevent it from configuring zero-copy/direct rendering so it is currently doing a CPU->GPU copy for the frame data, which is inefficient.

Using the `amcviddec-*` elements directly, or possibly `playbin`, should allow zero-copy to work properly. This requires including a few Java classes from the GStreamer development package in builds which is not a problem.

steveatinfincia, 2019-11-23:
Fixed in 464626210e92b5f2a4376df88f9316015646966d

closed 2019-11-23 by steveatinfincia

pilotnbr1, 2019-11-23:
Must be nice to finally close this!! nice job