OpenHD/QOpenHD #27: Air/Ground ATH9k_htc carrier sense medium access on QOpenhd

htcohio, 2019-11-01:
The RC uplink that is primarily used with atheros cards could benefit from having easy access to all of the other parameters in the link (other than TX power because that's already included in the main settings page).   

We would need fields within the QOpenhd app to define air and ground independently. 
.. then, SSync uploads the new parameters to the airside after reboot.

https://github.com/HD-Fpv/Open.HD_Image_Builder/blob/master/stages/05-Wifibroadcast/FILES/overlay/etc/modprobe.d/ath9k_hw.conf

steveatinfincia, 2019-11-02:
What would be needed is to add options for them in openhd-settings-1.txt, then have the ground scripts use them when running txpower_atheros (which is what alters the module parameter file up there at runtime).

Once we know what the settings are called in the openhd-settings-1.txt file I can add a mapping for them in the QOpenHD settings maps.

That would give us direct access to the parameters, however it may make more sense to have specific presets that group settings changes together. So instead of exposing all the individual parameters to users, there would just be ATH9K_PARAMETERS=residential or  ATH9K_PARAMETERS=field or whatever makes sense. Then the scripts would pick specific module parameters based on that setting, and the QOpenHD app would just have a dropdown with a few choices.

htcohio, 2019-11-06:
Hi,

I understand the idea to have "noisy" and "non-noisy" presets but, for the time being, it might be beneficial to just have the known ranges for each so we can easily adjust them....

So, for the CCA paramaters, The defaults are as follows:
options ath9k_hw txpower=58 aifs=2 cwmin=0 cwmax=3 cck_sifs=10 ofdm_sifs=16 slottime=9 thresh62=26

txpower (Already included)
aifs= (Range 0-5
cwmin= (Range 0-10)
cwmax= (Range 0-10)
cck_sifs= (Range 0-20)
ofdm_sifs=Range 0-50)
slottime= (Range 1-20)
Thresh62= (Range 20-62)

After some trial and error, perhaps we can define presets when we know what is best. 
It looks like we can use the same TXpower startup script from User1321 linked below:

https://github.com/HD-Fpv/Open.HD/commit/9dc39830d39f7c60544567444db38035023c570c
https://github.com/HD-Fpv/Open.HD/commit/198dcd4e228ae2372af3ee3d2d1257484739a8a2

(I will also copy this issue on the main Openhd Repo)

P.S. Here was Rodizios RCgroups recommendation for emiminating the CCA stuff for more bitrate.... Hoewever, Pi zero maxes out around 5-6mbps anyways so the best use case would be with a Pi2-3 air so you could use Datarate 2(11mbps) or 3(12Mbps) with higher video bitrate and long range.
https://github.com/rodizio1/EZ-WifiBroadcast/blob/develop/kernel/linux-4.9.28-ath9k_htc-misc.patch

steveatinfincia, 2019-11-09:
Added a mapping for all of them so they'll show up in the right spot with nice UI controls, the rest of the changes will have to be made in the Open.HD and builder repos

commit 8b1fccb7 (2019-11-09): Add ath_9k parameter mappings

Closes: #27
closed 2019-11-09 by steveatinfincia