OpenHD/QOpenHD #38: Custom Video Framerates

htcohio, 2019-11-16:
The Raspivid Framerate Options 30, 48 and 59.9 are the standard options however, some situations require custom framerates.

Examples:

CSI-HDMI adapter.... Some HDMI cameras only have PAL framerates of 24/25FPS.

IP Cameras at 30fps require 240FPS HelloVideo.bin player for the best playback

htcohio, 2019-11-16:
Also, 2 more notes:

1. Custom FPS is used experimenting with different Raspivid Slice params where Hello_video.bin 240 works better. 

2. Something not as obvious to the typical user/tester is that wfb video and player FPS are connected to some degree. 

I think I saw in hello_video that it defaults to 60fps output for most frame rates correct?

steveatinfincia, 2019-11-19:
> I think I saw in hello_video that it defaults to 60fps output for most frame rates correct?

Probably sync'd to the display refresh rate yea.

Is it enough to just make the UI control text-editable?

closed 2023-02-23 by Consti10