OpenHD/QOpenHD #41: Add ground station settings detail info for tooltips

steveatinfincia, 2019-11-23:
v0.3.0 added tooltips for each ground station setting, the information for them comes from the settings model/mapping. Each setting in the mapping can include an optional field named "info". The text in that field will automatically show up in the tooltip, and if a setting doesn't have one then "N/A" is currently shown.

The only settings that have an info field at the moment are `DISPLAY_OSD` and `ENABLE_QOPENHD`.

The tooltip area can be increased if it proves to be too small for the information we need to display there.

pilotnbr1, 2019-11-23:
Very nice! This makes the info block accessible to non coders

pilotnbr1, 2019-12-03:
The info field can be found at-

https://github.com/infincia/QOpenHD/blob/master/qml/ui/SettingsMap.qml#L20

pilotnbr1, 2019-12-15:
So if a tooltip does not have info I think it would be better to not display the "?"... Minor detail

closed 2023-02-23 by Consti10