OpenHD/QOpenHD #55: Settings are being saved even when they haven't been modified

steveatinfincia, 2020-01-06:
For some reason, ground pi settings are being marked as though they've been modified even when they haven't causing the settings system in the app to save them.

This isn't inherently a problem as long as the settings system is working properly, but it's not necessary either.

htcohio, 2020-01-06:
Hey,

So you're saying that the ground pi is periodically saving settings or, being triggered to save settings even when they haven't been modified?

That would explain why there are instances where Air & ground lose synchronization with settings even without changing anything.

this has not happened in recent test but in the past it has.

steveatinfincia, 2020-01-06:
I mean when you open the settings panel and change something like hotspot txpower, and then hit save, the app sends a bunch of other settings to the ground station to be saved as well even though the user hasn't changed them.

Because the other settings aren't actually changed it can't cause a problem with anything else (it's just telling the ground station to write the same value into the settings file that's already there).

commit 7dbe24b9 (2020-01-16): Don't save settings unless their value has changed

This replaces the previous simplistic code that just set a modified
property on the setting in the model and used that to detect which
settings had actually changed.

The problem is that for various reasons that flag gets set even when
the setting value isn't actually different from the original value,
presumably because the onValueChanged callback in QML gets triggered
for other reasons.

Instead we store the original setting value and check it during a save
which should prevent any setting from being saved unless it actually
changed.

Fixes [... cut]
closed 2020-01-16 by steveatinfincia