OpenHD/QOpenHD #56: Pi image - interface does not respond to touchscreen, mouse is working but pointer is invisible

msergiu80, 2020-01-10:
Hi,

First of all thank you for all your work. I installed the Pi 0.3.1 image and it is working great except 3 minor issues, one being the flight mode not working on the QOHD Osd and the other 2, maybe related, the touch events have no effect on the OSD and mouse works but pointer is invisible, so it is a hard guess of actual screen position to enter the menu and operate changes. My touchscreen is working otherwise (other SD with graphical OS) and also using the QOHD image, through SSH, I can see touch events with evtest, so touch drivers are installed and working. I did not change anything else from the default installation but if needed I can attach logs.

steveatinfincia, 2020-01-15:
Flight mode will be fixed in a few minutes in 0.3.2, the touchscreen issue I'm going to look at today.

As a workaround you can run QOpenHD on Android (or Windows/Mac) and connect to the ground station hotspot to use drag-and-drop remotely.

steveatinfincia, 2020-01-16:
Ok the touchscreen issue is fixed in [v0.3.3](https://github.com/OpenHD/QOpenHD/releases/tag/v0.3.3) and the nightly Open.HD image from tonight, [nightly-2020-1-15](https://github.com/OpenHD/Open.HD/releases/tag/nightly-2020-1-15).

I've created another issue for the mouse pointer, #58.

closed 2020-01-16 by steveatinfincia