OpenHD/QOpenHD #87: Use consistent coding style

steveatinfincia, 2020-04-02:
A lot of the code has mixedCase and underscore_case, which is inconsistent but also leads to odd situations where Qt derives the name of a slot automatically, and ends up with `onSlot_nameChanged`.

Qt seems to prefer mixed case, so we should just use it everywhere.

We should also ensure there are 2 spaces between method implementations, and use the /* */ comment style for multi-line comments.

pilotnbr1, 2020-04-02:
> A lot of the code has mixedCase and underscore_case, which is inconsistent but also leads to odd situations where Qt derives the name of a slot automatically, and ends up with `onSlot_nameChanged`. [...]

Noted! And guilty :p ... Ill try to some cleanup

steveatinfincia, 2020-04-02:
Nah it's me, I've been using different case styles every other day :D

closed 2022-11-07 by raphaelscholle