OpenHD/QOpenHD #103: 3D stereoscopic viewmode needed (StereoPi)

careyer, 2020-04-15:
**Introduction / Motivation**
Thanks to the support for the CM3 (and thus the StereoPi) it would be very useful to have a 3D stereoscopic viewmode for both:
- QOpenHD App ..and..
- QOpenHD GUI running on the GndPi. 

The 3D effect in the side-by-side video can already by leveraged viewing it either:
- using Google cardboard style 3D-googles (phone)
- or connecting the HDMI to dedictaed FPV / VR goggles which can split the side-by-side image and display the resulting half-image for the left/right eye respectively (e.g. Zeiss Cinemizer)

Only drawback in the current state is that the OSD needs to be deactivated right now, beause it does not follow the side-by-side splitvideo mantra but gets overlayed accross the whole width of the scene. 

**What is needed?**
Display mode mode that shows OSD also in a side-by-side manner (i.e. twice - overlayed over each of the two side-by-side half-images)

I hope this can be solved with Qt in an elegant way?

steveatinfincia, 2020-04-16:
Yep that can be done :)

Are you thinking of retaining access to the interface via touch, or just for OSD purposes?

Non-touch would be trivial, just make the OSD width half the screen, and copy the pixels to the other side.

careyer, 2020-04-16:
I am happy to hear this is indeed feasable.

I think there are 4 possible solutions:

a) dispable Touch and use duplicate OSD 
b) keep Touch enabled and allow touch input on either of the two half pictures
c) on GndPi display only one half image and a single OSD on the 7" screen (DSI)  and on the HDMI output both half pictures side-by-side with duplicated OSD
d) add a function to quickly toggle between "Mono" and "Stereo" Mode. So if Touch input is needed toogle to Mono mode... perform adjustments ...then switch back to Stereo Mode...

I think option d) might be the most practical solution?

realizator, 2020-04-23:
It will be great to add an option for relative position of duplicated OSDs (X axis), as in stereoscopic view you'll get 3D effect and will "feel" the distance for the OSD.

p.s. If you have any StereoPi-related questions, I can help with this too, as I'm one of the developers.

steveatinfincia, 2020-05-31:
After talking to @careyer about this on telegram the solution is actually much simpler, we just need a button that splits + duplicates the HUD layer on the screen, since nobody is going to be touching the screen at that point the touch interaction doesn't really matter so we don't have to do anything special with that.

steveatinfincia, 2020-05-31:
> It will be great to add an option for relative position of duplicated OSDs (X axis), as in stereoscopic view you'll get 3D effect and will "feel" the distance for the OSD.

Good idea :)

> p.s. If you have any StereoPi-related questions, I can help with this too, as I'm one of the developers.

Hey, nice to meet you :) Wonderful job on the stereo pi hardware!

steveatinfincia, 2020-07-10:
The big parts are implemented now, using an OpenGL shader to clone the HUD area in realtime using the GPU.

We'll have to give some thought to handling the widget layout in 3D mode/

steveatinfincia, 2020-07-10:
[image]

Crowded at the moment, scaling should fix 90% of that.