OpenHD/QOpenHD #118: no home position in qopenhd

kasatka60, 2020-06-05:
inav 2.4
mavlink telemetry.
qopenhd v0.4.0b4 windows 10 x64
qopenhd v0.4.0b4 android
qopenhd 2.0.0rc6
after arming writes: "no home position in openhd"

pilotnbr1, 2020-06-07:
Do you have a "good" 3d gps fix before arming?

Does an "H" marker appear on the map at any time?

This might be a ardupilot vs inav difference.. Currently the home position is sent from the flight controller via mavlink message. It is no longer guessed by the osd

kasatka60, 2020-06-07:
Satellites 11-12
marker "H" always has coordinates 0
May add guessing home point in qopenhd?

pilotnbr1, 2020-06-07:
> May add guessing home point in qopenhd?

The "old" osd would determine home position purely on the gps position data from the flight controller

qopenhd reads MSG_ID_HOME_POSITION from the flight controller which is the positions that the flight controller thinks home is.

kasatka60, 2020-06-07:
it's a shame

steveatinfincia, 2020-06-07:
I suspect that either the mavlink message is never arriving, or it's arriving *after* the arming happens, so the message about not having a home position is always shown.

Someone else reported the home distance not working too, I can look into it if Luke hasn't already.

pilotnbr1, 2020-06-08:
> it's a shame

Dont worry! We will figure this out! :)

In ardupilot I can see the home position is ony getting sent once- when the flight controller actually sets it. If I connect qopenhd to the fc after that event I also will not get a home position- so the mavlink home position message is not being constantly sent ( maybe a stream rate issue here too, but I think not). Ardupilot takes some time to set the position after fc start so it is easy to catch

For Inav I suspect that we are just missing the moment when it is sent. I imagine you fc boots and gets a 3d gps fix much faster than openhd on the pi. 

For testing AND as a temporary fix I see that in inav you can reset the FC home position with your transmitter (maybe mission controller too)- Can you please try this

For permenantly fixing this maybe we need to request the home position via mavlink from the fc upon arm or find some other message that always broadcast

kasatka60, 2020-06-09:
Today I will try

kasatka60, 2020-06-09:
Checked and does not work.
Inav does not send MSG_ID_HOME_POSITION

So how to bring rssi on osd?
Also, telemetry from raspberry pi to the u360 does not work. Or I messed up something.

pilotnbr1, 2020-06-09:
> Checked and does not work.
> Inav does not send MSG_ID_HOME_POSITION
> 
> So how to bring rssi on osd?

Thanks for testing!

Tomorrow I will switch qopenhd homeposition stuff to do what the old osd did- determine home position in the app.

At some future time will revisit getting it from fc

Rssi for regular rc display in qopenhd has not been done yet. I will add it on my list. Probably do it before end of the month.

kasatka60, 2020-06-09:
can I still help testing somehow?

pilotnbr1, 2020-06-09:
> can I still help testing somehow?

I dont think any of us will turn away help with testing! 😀

There wont be much to test until Stephen rolls out rc7... Not sure when that will be exactly

steveatinfincia, 2020-06-14:
Should be fixed in `0.4.0b5`

closed 2020-06-14 by steveatinfincia