OpenHD/QOpenHD #159: Android device as a ground station

questioner314, 2020-09-22:
Is it possible to use an Android device as a raspberry-free ground station? The computing power of Android tablets PC and smartphones is higher than that of raspberries. Connecting the WIFI receiver to the USB host of a tablet pc or phone would be a more compact and cheaper solution. It would also be possible to use HDMI android TV stiks with monitors.

steveatinfincia, 2020-09-24:
Probably not android, it's generally not worth the mess of missing drivers and old kernels with different bugs.

We will support x86 tablets though, because we can run the full image on them and ensure everything works.

cgrrty, 2021-01-29:
When does the ground station just require an x86 pc, I am very interested in it

Consti10, 2023-02-06:
Possible in evo

closed 2023-02-06 by Consti10