OpenHD/QOpenHD #204: Release 2.0.10 blank screen

pilotnbr1, 2021-07-05:
On pi3b+ as groundpi Mario gets a blank screen.

He uses adsb widget on opensky connected to a phone via hotspot for data.

Not sure if the whole groundpi crashes or just the app.

white-rabbitGR, 2021-07-08:
I tested my SDR receiver (NooElec Nano 2+) on 2.0.8 and it works perfectly.
Then, I updated the qopenhd via ssh and it started chasing crashing with blank screen. 
Maybe use as baseline the same version of 2.0.8 and simply add the Virtual Reality feature.

pilotnbr1, 2021-07-13:
Mario config-

When I enabled data from SDR in the ADS-B widget it crashed. 
I did the test in both 720p or 1080p. 
When I disable completely the data from SDR receiver everything is fine. 
I also wanted to overrule any hardware interferences so I disabled the antenna tracker and I connected directly my Fatshark HDO goggles to the RPi3B+. 
I didn’t turn on the UAV. 
I didn’t have internet data. 
I didn’t have the UAV connected. 
I am using the NooElec Nano 2+ SDR receiver.

white-rabbitGR, 2021-07-13:
In Pi4B the SDR is working fine. 
Thank you for all your hard work! 
Cheers mate!😊🍻

pilotnbr1, 2021-07-19:
A logger class has been added that will now write whatever debug statements we want to /boot on the pi. Once deployed, hope it sheds more light on where and how sdr is breaking

pilotnbr1, 2021-07-22:
More and more this is looking like memory issues.

SDR has no range limit and often will "see" aircraft up to 200km away. For people in high traffic areas this results in alot of airplane traffic being drawn

David has pointed out that we are probably not drawing plane icons on the map as efficiently as we could. He referenced the sae issue in QGC 
https://github.com/mavlink/qgroundcontrol/pull/9513
https://github.com/mavlink/qgroundcontrol/issues/9501

This could also have implication for the mission widget

pilotnbr1, 2021-08-29:
By restricting the data going into the model we are avoiding crashes on the pi

closed 2021-08-29 by pilotnbr1