OpenHD/QOpenHD #215: We Should Request_data_stream

pilotnbr1, 2021-07-15:
For easier configuration and sync between qopenhd app and fc we should Request_data_stream

pilotnbr1, 2021-07-18:
Stephen already did this. Might not work for px4...

case MAVLINK_MSG_ID_SYSTEM_TIME:{
            mavlink_system_time_t sys_time;
            mavlink_msg_system_time_decode(&msg, &sys_time);
            uint32_t boot_time = sys_time.time_boot_ms;

            if (boot_time < m_last_boot || m_last_boot == 0) {
                m_last_boot = boot_time;

                setDataStreamRate(MAV_DATA_STREAM_EXTENDED_STATUS, 2);
                setDataStreamRate(MAV_DATA_STREAM_EXTRA1, 10);
                setDataStreamRate(MAV_DATA_STREAM_EXTRA2, 5);
                setDataStreamRate(MAV_DATA_STREAM_EXTRA3, 3);
                setDataStreamRate(MAV_DATA_STREAM_POSITION, 3);
                setDataStreamRate(MAV_DATA_STREAM_RAW_SENSORS, 2);
                setDataStreamRate(MAV_DATA_STREAM_RC_CHANNELS, 2);
            }

            break;
        }

pilotnbr1, 2021-07-19:
There is a new (as of arupilot 4.0) way of setting precise message flow. https://ardupilot.org/dev/docs/mavlink-basics.html

Need to look at this closer

Consti10, 2023-07-27:
Will be fixed in the next release:
https://github.com/OpenHD/QOpenHD/blob/2.4.2-beta/app/telemetry/models/fcmessageintervalhelper.hpp#L94

Note that we use the newest / recommended, aka use MAV_CMD_SET_MESSAGE_INTERVAL

closed 2023-07-27 by Consti10