OpenHD/QOpenHD #305: Migrate qopenhd from qt 5 to 6

pilotnbr1, 2022-05-15:
Qt 5.15 is getting old and now looking into migrating qopenhd to Qt 6.X

PROS:
- qmake to cmake
- get qt shader tools. But no present plans to use them

CONS:
- more work... some of it will be platform specific. Nobody currently debugs all platforms like Stephen
- mapbox-gl is not a supported plugin. MaplibreGL looks to be the new solution

Current errors seen while upgrading:
1) Mapbox plugin related
2) OSM plugin for maps error probably related to Mapbox
3) QT x11extras (https://doc.qt.io/qt-6/extras-changes-qt6.html) solution will be somewhat platform dependant
4) QT Location - no plans to port.. Must switch to QT positioning
5) QT gamepads depreciated. probably not a big deal as we use 3rd part qt joystick But need to remove gamepad code
5) QT android extras will probably need some attention for android
6) QT openGL will need attention most likely
7) QT speech supposedly being ported in qt 6.4
8) Qt Windows Extras might need attention for windows

pilotnbr1, 2022-05-16:
Maplibre is not really targetting qml. The efforts to get it to work on QML have required QT location. Qt Location is also depreciated in QT 6. So the only working Maplibre does a work around using "MSVC" and the performance sounds like its bad https://github.com/ntadej/qt-geoservices-maplibre

Not having mapping is deal breaker for me while weighing 5.15 vs 6.x . As of now does not seem worth the effort to switch.

pilotnbr1, 2022-05-18:
ArcGIS Runtime API for Qt might be an option for mapping in qt6. We would be getting back into needing an api key/token . Not clear on how that distributes when its deployed in their "essentials" plan

pilotnbr1, 2022-05-18:
QT 6.2 seems to be getting some map support using maplibre as a plugin. Not sure about linux support.
https://bugreports.qt.io/plugins/servlet/mobile#issue/QTBUG-96795

To try it need QtPositioning and QtShaderTools which are not installed by default

pilotnbr1, 2022-05-18:
qt 5 works with cmake. So not a problem switching away from qmake

raphaelscholle, 2022-06-22:
We updated to 5.15.4, so I think with all it's problems we're currently not interested in updating to 6.x

raphaelscholle, 2026-10-20:
﻿Status update for **dev-release**:

QOpenHD already has a Qt 6 build path. The root [CMakeLists.txt](https://github.com/OpenHD/QOpenHD/blob/dev-release/CMakeLists.txt) targets Qt 6.4 or newer, and the tree includes Qt-version-specific implementations and an [Android Qt 6 build workflow](https://github.com/OpenHD/QOpenHD/blob/dev-release/.github/workflows/build_android_signed_qt6.yml). That workflow currently triggers on 2.7-evo, so its presence should not be confused with an automatically verified dev-release Qt 6 release.

The remaining question is platform/backend readiness and deployment, rather than starting a Qt 6 port from scratch. Our existing Raspberry Pi deployment retains Qt 5 with its legacy FKMS/video-overlay integration. Moving that deployment to Qt 6 needs the complete display/video path validated together. Qt 6 itself supports Raspberry Pi through modern EGLFS/KMS/GBM; the Pi is not generally unsupported by Qt 6. See [Qt's embedded-device configuration documentation](https://doc.qt.io/qt-6/configure-linux-device.html). I would therefore describe the constraint as our current Pi display/video stack, rather than a blanket statement that all FKMS configurations cannot work with Qt 6.

Full Qt 6 feature parity also should not be inferred from the build files alone: the CMake path currently leaves the avcodec, Android-video and GStreamer video includes commented out. Platform-specific video and mapping behavior still need their own verification.

Regarding Qt 5 maintenance: **we have an enterprise/commercial Qt license**, but the support distinction matters. Qt 5.15's standard commercial support ended on **26 May 2025**, and Qt 5.15.19 was the last regular commercial LTS release. Qt continues to offer **Extended Security Maintenance (ESM)** for Qt 5.15 and separate Extended Support services. ESM is an additional subscription covering critical vulnerability/security patches for most Qt libraries; an enterprise license by itself does not prove ESM coverage. This comment does not assert that our contract includes that additional subscription.

Official Qt references:
- [Qt 5.15.19 release and end of standard support](https://www.qt.io/blog/commercial-lts-qt-5.15.19-released)
- [Current legacy Qt security-maintenance offering](https://www.qt.io/development/qt-framework/security-maintenance)

So the original issue now needs to distinguish existing Qt 6 support from completing/validating each platform migration, particularly the Raspberry Pi video/display path. No new full-application or Pi hardware validation is being claimed by this status update.

raphaelscholle, 2026-10-20:
﻿Closing this migration issue with the current platform policy:

- QOpenHD already has a Qt 6 build path in **dev-release**. Qt 6 work and platform-specific validation can continue on other platforms.
- Our existing Raspberry Pi platforms will remain on **Qt 5 with FKMS**. We have **no plans to migrate those images to KMS or Qt 6**; retaining this stack is a deliberate platform decision.
- We may reconsider if future Raspberry Pi hardware restores hardware video encoding and decoding capabilities that justify revisiting the display/video stack. For now, no such migration work is planned.

This closes the broad Qt 5-to-6 migration tracker. It does not claim that every Qt 6 video/mapping backend has full feature parity or has been hardware-validated. Concrete issues on supported Qt 6 targets can be handled separately against dev-release.

The related Qt 6 proposal #858 is also being closed without merging. See the preceding status comment for the commercial Qt 5 maintenance/ESM distinction; this policy does not change that distinction.

closed 2026-10-20 by raphaelscholle