OpenHD/QOpenHD #350: invalid conversion from ‘AVCodec**’ to ‘const AVCodec**’

Zipdox, 2022-10-30:
On the 2.2.3-evo-consti-test branch.
```
../app/exp/avcodec_decoder.cpp:487:69: error: invalid conversion from ‘AVCodec**’ to ‘const AVCodec**’ [-fpermissive]
  487 |     ret = av_find_best_stream(input_ctx, AVMEDIA_TYPE_VIDEO, -1, -1,(AVCodec**) &decoder, 0);
      |                                                                     ^~~~~~~~~~~~~~~~~~~~
      |                                                                     |
      |                                                                     AVCodec**
      ```

Consti10, 2022-11-03:
yeah, some ffmpeg installations require "(const AVCodec**)", and some require (AVCodec**).
There is no better way than to fix this manually according to what ffmpeg you get on you system.

closed 2022-11-03 by Consti10

Zipdox, 2022-11-03:
Is it version related? I'm sure there's a macro you can use to deal with it

Consti10, 2022-11-04:
I am pretty sure it is, you can also find it on stackoverflow - but no proper solution(s) as of yet

Zipdox, 2022-11-15:
[image]