OpenHD/QOpenHD #454: display not connect on RPI

karu2003, 2023-02-01:
how to build QOpenHD on raspberry so as not to get a display error?

qt.qpa.xcb: could not connect to display
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

I run so sudo QT_QPA_PLATFORM=eglfs ./QOpenHD

raphaelscholle, 2023-02-01:
first edit /boot/config.txt from dtoverlay=vc4-kms-v3d to dtoverlay=vc4-fkms-v3d 

sudo apt update
sudo apt -y install curl git
curl -1sLf 'https://dl.cloudsmith.io/public/openhd/openhd-2-3-evo/setup.deb.sh' | sudo -E bash
sudo apt -y upgrade
sudo apt -y install dkms openhd qopenhd
sudo apt -y install libqt5concurrent5 libqt5core5a libqt5dbus5 libqt5designer5 libqt5gui5 libqt5help5 libqt5location5 libqt5location5-plugins libqt5multimedia5 libqt5multimedia5-plugins libqt5multimediagsttools5 libqt5multimediawidgets5 libqt5network5 libqt5opengl5 libqt5opengl5-dev libqt5positioning5 libqt5positioning5-plugins libqt5positioningquick5 libqt5printsupport5 libqt5qml5 libqt5quick5 libqt5quickparticles5 libqt5quickshapes5 libqt5quicktest5 libqt5quickwidgets5 libqt5sensors5 libqt5sql5 libqt5sql5-sqlite libqt5svg5 libqt5test5 libqt5webchannel5 libqt5webkit5 libqt5widgets5 libqt5x11extras5 libqt5xml5 openshot-qt python3-pyqt5 python3-pyqt5.qtopengl python3-pyqt5.qtsvg python3-pyqt5.qtwebkit python3-pyqtgraph 
sudo apt -y install qml-module-qt-labs-settings qml-module-qtgraphicaleffects qml-module-qtlocation qml-module-qtpositioning qml-module-qtquick-controls qml-module-qtquick-dialogs qml-module-qtquick-extras qml-module-qtquick-layouts qml-module-qtquick-privatewidgets qml-module-qtquick-shapes qml-module-qtquick-window2 qml-module-qtquick2 qt5-gtk-platformtheme qt5-qmake qt5-qmake-bin qt5-qmltooling-plugins qtbase5-dev qtbase5-dev-tools qtchooser qtdeclarative5-dev qtdeclarative5-dev-tools qtpositioning5-dev qttranslations5-l10n
sudo apt -y install libqt5texttospeech5-dev raspberrypi-kernel-headers

this is compiling our 8812au driver into the standart-kernel (only works for 8812au [obviously])

cd ~
git clone https://github.com/OpenHD/rtl8812au
cd rtl8812au
sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
sed -i 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile
ln -s /usr/src/linux-headers-$(uname -r)/arch/arm /usr/src/linux-headers-$(uname -r)/arch/$(uname -m)
sudo ./dkms-install.sh
sudo touch /boot/openhd/ground.txt
sudo reboot

now you can start QOpenHD, within a terminal on the raspberry desktop (ssh wont work) 

QOpenHD --platform xcb

raphaelscholle, 2023-02-01:
btw. don't expect it to have any sort of hardware acceleration, there may be a chance that it can't even display the video at all

karu2003, 2023-02-02:
thanks. I'm trying to build 2.3-evo directly on RPI. OpenHD image. After building i run /build/release/QOpenHD --platform EGLFS. I would like to have it by default.

raphaelscholle, 2023-02-02:
you can just build your own .deb package and install it via dpkg -i that'll make it automatically start

	    sudo apt install -y git libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-good
            sudo apt install -y openhd-qt 
            sudo bash install_dep_rpi.sh
            sudo ./package.sh armhf raspbian bullseye

closed 2023-02-06 by Consti10