OpenHD/QOpenHD #473: Efficiency mah / km value reported doesn't work

Consti10, 2023-02-12:
This comes from the fact we renamed some of the battery related variables, and the factt we calculated it at run time in QOpenHD.
We have to either document the way it was calculated & add it back in or check if there is a mavlink message for us that can report it.

Consti10, 2023-02-25:
Clarification:

The mAh value refers to the total current consumed.
It uses mavlink_battery_status_t 's
`int32_t current_consumed; /*< [mAh] Consumed charge, -1: autopilot does not provide consumption estimate*/`
value - This value should be correctly displayed in QOpenHD.

The "efficiency" in mAh/km is known to be not working, there is no mavlink message for that and calculating it is not easy.

Flierman1945, 2023-02-28:
Consti, With ref to your clarification above, I am a little confused.
Are you saying there is not a problem with "mah consumed" and it should display correctly. 
When I ran ver 2.0.12 it all worked ok bar the mah/km .
Also, your beta hotfix:- is it the case you have only released the update files rather than a complete image.  Can you give me any indication when the beta complete image will be available.

Consti10, 2023-02-28:
The mAh value should work - it worked in the last test flight from norbert I saw and includes no additional calculation(s) -
aka your fc reports the mAh (charge consumed) and QOpenHD displays that value.

Efficiency is a bit more complicated, it needs to be calculated in QOpenHD taking delta distance / delta charge into account.
This code has not been ported yet / needs to be re-done. 

The beta hotfix is the one you get from the image writer (tagged in github)
The next release will be [v2.3.2-alpha](https://github.com/OpenHD/QOpenHD/releases/tag/v2.3.2-alpha)
I tagged it already, but then we found a couple of bugs (mostly library / image related) which meant we had to delay things.
R.n the schedule is:
Next friday -> beta tester group
If things work out, we'l update the .img / the image on the image builder on Saturday / Sunday.

Flierman1945, 2023-02-28:
Thx for the release info.
I will need to go back over all my settings to see if I have missed any thing ref total mah consumed

knputyi, 2023-02-28:
Exactly, total consumed mah is working in EVO, only mah/km is not calculated. @Flierman1945 , it should work if using mavlink.

Consti10, 2023-03-15:
I've added the most simple form of calculating the efficiency during flight I could come up with.
Can't really test it without a flight, but regardless, better than nothing

closed 2023-04-22 by Consti10