OpenHD/QOpenHD #528: Home Position not set when using INAV Mavlink telemetry

tipoman9, 2023-04-05:
Respectively, Distance to home and Home vector do not work in flight.
qOpenHD needs a MAVLINK message of  type:
  case MAVLINK_MSG_ID_HOME_POSITION:{....}
to set home position. 
I checked INAV implementation of mavlink protocol, there is imported definition for :
static inline uint16_t mavlink_msg_home_position_pack( .... ) 
type of message,  but it is never called. 
Seems INAV only sends mavlink_msg_gps_raw_int_pack messages.
Which means that home vector and distance to home will never work with INAV.
I see two options - change INAV to implement this type of message.
Or think out some clever way for qOpenHD to get home position.
Either accept the last reported coordinates when an "Armed" message is received from INAV, or if this is not reliable, add the possibility for the user to accept current coordinates as home ones by clicking on the Home Distance indicator for example.

Consti10, 2023-04-05:
As a temporary solution:
https://github.com/OpenHD/QOpenHD/commit/69c04df4941a53ded369e1653a351c36242ed47c

Consti10, 2023-04-05:
use last reported coordinates when an "Armed" message is received

Consti10, 2023-04-05:
or rather, uses the first global position message after the fc is armed to set home position

Consti10, 2023-04-06:
Since it is already merged & built by the CIs, you can probably just apt update / upgrade.
Please report back if it "works".
inav quirks can be enabled under the QOpenHD dev registry

tipoman9, 2023-05-21:
I confirm it works. Tested it this week several times, works flawlessly.

closed 2023-05-23 by tipoman9