OpenHD/QOpenHD #530: Error: "Could not initialize window system"

yuace, 2023-04-06:
I use qt+eglfs in qualcomm arm64 board, it's can's support avcodec. So I want to use gstreamer to decode and display the video.
 1. comment _#include(app/videostreaming/avcodec/avcodec_video.pri)_
 2. enable _DEFINES += QOPENHD_GSTREAMER_PRIMARY_VIDEO_
 
But meet error: Error: "Could not initialize window system".

Could u give some advice about it.

Consti10, 2023-04-06:
can't or can ?

You can 
1) try force sw decode - that should work on every platform
2) try compiling with all video stuff disabled, then make a transparent surface & your own video display.

2) often doesn't work though (kms master)

yuace, 2023-04-06:
收到！！！

Consti10, 2023-04-06:
also, to launch, pass --platform eglfs or something (that then depends on your device)

yuace, 2023-04-06:
yes, I use default eglfs and sw avcodec can work, but I want transfer 1080p 60fps, but qualcomm gpu don't support avcodec api. So I want to try gstreamer pipeline.

And i debug gstreamer find that:
 qtglwidget qtitem.cc:335:initWinSys:[00m 0x55b33da460 failed to retrieve wrapped context (NULL)
0:00:04.260466230 [333m46394[00m   0x55af44f200 [33;01mWARN   [00m [00m              qtsink gstqtsink.cc:307:gst_qt_sink_change_state:<qmlglsink>[00m error: Could not initialize window system
 GST_ERROR_SYSTEM gstelement.c:2153:gst_element_message_full_with_details:<qmlglsink>[00m posting message: Could not initialize window system

Consti10, 2023-04-06:
For something as complicated as this, I'd recommend going with qopenhd transparent and manually launching the appropriate gstreamer pipeline (opening a window behind).

If you are lucky, that works.

yuace, 2023-04-06:
I get you mean, but if I run no window linux that can I use gstreamer pipeline display such as waylandsink?

Consti10, 2023-04-06:
On rpi, this works with a couple of Quirks on fkms, and on rk3588, something similar works too (I think there it is kmssink).

But yes, if you are unlucky, you have to completely write your own display pipeline / Quirks for qualcom. Perhaps using qt and opengl (the current source code in qopenhd might help, but things like that are incredibly difficult).

You can also use rk3588 though, there we plan to add officiall ground station support for.
At least the radxca board I think it was.

yuace, 2023-04-06:
OK, thanks a lot. Actually I focus on wifi, don't know much about graphics and displays.

I now foucs on qualcomm QRB5165, it's support wayland, so can I run qopenhd on wayland and how can i do it?

Consti10, 2023-04-06:
With a window Manager like wayland, you can just run qopenhd in a window and the video in a window surface behind it.
Not necessarily low latency, but possible.
I can't help you with the (gstreamer) Pipelines though, since I do not have your exact board.

yuace, 2023-04-07:
I try with wayland,but still meet some issues, Failed to create wl_display(No such file or directory).
I don't know if I compile the QT5.15.X with wrong parameters.

raphaelscholle, 2023-04-15:
Compiling QT is a pretty intense process and is individual to the SBC you're using, if you want you can look into our build.sh 
 
https://github.com/OpenHD/OpenHD-Qt/blob/2.2.3-evo/build.sh

that's including some settings I use for different platforms.
Keep in mind that HW-acceleration and EGLFS need special libraries and may be different on your board.
This is normally handled in the "mkspecs"

In general we recommend having at least a linux kernel 5.x since this normally enables you using a distribution which has working QT15.x