OpenHD/QOpenHD #584: QOpenHD crashes on ArduPilot MavLink message

entertain2017, 2023-07-16:
Hello,

QOpenHD keeps crashing on the Ground Station.

QOpenHD: app/telemetry/models/fcmavlinksystem.cpp:38: void FCMavlinkSystem::set_system(std::shared_ptr<mavsdk::System>): Assertion `m_system==nullptr' failed.

my setup.RPI4 (Air and Ground) RC_JOY enabled (also crash if disabled).. Uart on Serial0 to a speedybee f405wing, with ardupilot FW.. This does not happen with the INAV Firmware...so it seems a Ardupilot MavLink message is crashing the QOpenHD.

OpenHD i have tried 2.4.1-evo and the 2.4.evo beta... ArduPilot is the Arduplane4.3.5_SpeedyBeeF405WING_withparm_bl.hex from the Speedybee Website.

Consti10, 2023-07-18:
Hm, weird. There are probably 2 mavlink systems reporting themselves, and only one of them is the autopilot.
And qopenhd is having trouble figuring out who is who - and sets them both as autopilot 
(Which then fails at the assertion, there should only ever be one autopilot aka fcmavlinksystem.
Any idea what's special about your setup ? Did you change sys ID, have another mavlink system on the air ?

Consti10, 2023-07-18:
Note that on a classical setup, there are 3 mavlink systems discoverable from the ground -
The openhd ground unit, the openhd air unit and the fc itself.
You seem to have another system somewhere ... ?

Consti10, 2023-07-18:
The root cause of the problem is here 
https://github.com/OpenHD/QOpenHD/blob/2.3-evo-consti-dev/app/telemetry/MavlinkTelemetry.cpp#L80

entertain2017, 2023-07-24:
thats the funny about it.. my setup is simple as it could be.. Ground. Air and FC. I also tried to set a other sysid on the FC. no luck..

Consti10, 2023-07-24:
What fc are you using?

Consti10, 2023-07-24:
Sorry, ardupilot. That is really weird, a lot of people use ardupilot successfully - including the speedybee.
I did / will add some stuff in the next release to at least not crash, but only log a warning in this case

romarysonrier, 2023-08-29:
Confirmed for me :

> systemctl status qopenhd => (code=killed, signal=ABRT)

> May 14 04:25:05 openhd QOpenHD[5486]: FCMavlinkSystem::set_system: FC SYS ID is: 1
> May 14 04:25:05 openhd QOpenHD[5486]: MAVSDK:: COMMAND_LONG 511 to send to 1, 1
> May 14 04:25:05 openhd QOpenHD[5486]: System found 255  with components: 1
> May 14 04:25:05 openhd QOpenHD[5486]: Component: 230
> May 14 04:25:05 openhd QOpenHD[5486]: Found FC
> May 14 04:25:05 openhd QOpenHD[5486]: QOpenHD: app/telemetry/models/fcmavlinksystem.cpp:38: void FCMavlinkSystem::set_system(std::shared_ptr<mavsdk::System>): Assertion `m_system==nullptr' failed.
> May 14 04:25:05 openhd systemd[1]: qopenhd.service: Main process exited, code=killed, status=6/ABRT
> May 14 04:25:05 openhd systemd[1]: qopenhd.service: Failed with result 'signal'

Consti10, 2023-11-24:
should be fixed on the upcoming release

closed 2023-11-24 by Consti10