OpenHD/QOpenHD #593: Can't install on rpi (errors /var/cache/apt/archives/gst-openhd-plugins_2.3-evo-03031839_armhf.deb)

krzysztofkuczek, 2023-09-14:
HI,
After 
` curl -1sLf   'https://dl.cloudsmith.io/public/openhd/openhd-2-3-evo/setup.deb.sh'   | sudo -E bash`

I try to install, without success
```
 sudo apt-get install qopenhd
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  gst-openhd-plugins
The following NEW packages will be installed:
  gst-openhd-plugins qopenhd
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/17.6 MB of archives.
After this operation, 43.2 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
(Reading database ... 278487 files and directories currently installed.)
Preparing to unpack .../gst-openhd-plugins_2.3-evo-03031839_armhf.deb ...
Unpacking gst-openhd-plugins (2.3-evo-03031839) ...
dpkg: error processing archive /var/cache/apt/archives/gst-openhd-plugins_2.3-evo-03031839_armhf.deb (--unpack):
 trying to overwrite '/usr/share/gstreamer-1.0/presets/GstIirEqualizer10Bands.prs', which is also in package gstreamer1.0-plugins-good:armhf 1.18.4-2+deb11u2+rpt1
Selecting previously unselected package qopenhd.
Preparing to unpack .../qopenhd_2.5-evo-202309130127-d41f457b_armhf.deb ...
Unpacking qopenhd (2.5-evo-202309130127-d41f457b) ...
Errors were encountered while processing:
 /var/cache/apt/archives/gst-openhd-plugins_2.3-evo-03031839_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

```
Any hints?
Thank you in advance
Kris

PS
I was also trying to compile but run out of memory :(
`cc1plus: out of memory allocating 134217704 bytes after a total of 90972160 bytes`

Consti10, 2023-09-20:
You can specify --force-overwrite or similar to install the package anyways

Consti10, 2023-09-20:
@oom: if you want to do any development, treat yourself a rpi 4 ;) (I assume you are on rpi)

closed 2023-09-20 by Consti10