OpenHD/QOpenHD #595: QOpenHD for rock5

xueshenan, 2023-09-25:
Hi , 
     I want to use QOpenHD in rock5(rk3588 platform). I found the project has [rock5_h264_decode.service](https://github.com/OpenHD/QOpenHD/blob/2.5-evo/systemd/rock5_h264_decode.service). So how to use the rock5 decode service in **QOpenHD**.

xueshenan, 2023-09-26:
For now I known how to use **rock5_h264_decode.service** to start video streaming service. But one more question, how to use the **[rock5_qopenhd.service](https://github.com/OpenHD/QOpenHD/blob/2.5-evo/systemd/rock5_qopenhd.service)**, I just compile the QOpenHD and lunch the **rock5_qopenhd** service, but the QOPenHD is lunch failed.

raphaelscholle, 2023-09-26:
Getting hardware accelerated decoding is extremely tricky, in the next OpenHD Release it will be included. On the current images it doesn't work and just installing it on a prebuild image will also not work.

If you want to use the Rock5 as ground you should be able to just do it with one of the images we provide for the platform.

xueshenan, 2023-09-27:
For now, I know how to lunch QOpenHD in rock5 with **eglfs**. I want to use the hardware decoder, so I need to start  rock5_h264_decode.service as video layer and the [rock5_qopenhd.service](https://github.com/OpenHD/QOpenHD/blob/2.5-evo/systemd/rock5_qopenhd.service) as ui layer.

```
__________________________
|              ___________|_______________
|              |                          |
|  video       |                          |
|  layer       |                          |
|              |    ui  layer             |
|              |                          |
|______________|                          |
               |__________________________|
```

the service can lunch success, but the ui layer is not render transparent, so the video layer is not visiable.

closed 2023-09-27 by xueshenan