OpenHD/QOpenHD #609: Custom Widget & Mavlink

AlexNester-QuadX, 2023-11-18:
I am creating a custom widget that will use data from mavlink.
So facing some problems:

1. How connect to QopenHD using pymavlink via udp? (Pymavlink on companion computer)

I was try : (https://mavlink.io/en/mavgen_python/)

```
from pymavlink import mavutil

#Start a connection listening on a UDP port_
the_connection = mavutil.mavlink_connection('udpin:localhost:14550') 

#Wait for the first heartbeat_ 
#his sets the system and component ID of remote system for the link_
the_connection.wait_heartbeat()
```

was trying addresses but nothing
('udpout:0.0.0.0:14550') and ('udpin:0.0.0.0:14550') 
('udpout:192.168.2.1:5760') and ('udpin:192.168.2.1:5760') 

and this tutorial not helped me https://www.ardusub.com/developers/pymavlink.html

2. How it possible send own messages? I noticed that all widgets use a class _fcMavlinkSystem

it will be great add to Class 2 objects for example 

    L_RO_PROP(int,detect_x,set_detect_x,0)
    L_RO_PROP(int,detect_y,set_detect_y,0)

[image]

and then use in my widget

[image]

But hove I can send this from companion computer whith Pymavlink or dronekit?

May be better way use Define MAVLink Messages & Enums ? (https://mavlink.io/en/guide/define_xml_element.html)

please nudge me in the right direction

skol101, 2024-01-20:
Seconded, also intersted in adding custom layer/widget to OSD on the GCS.

raphaelscholle, 2026-10-20:
﻿Documenting the answer for the current **dev-release** branch and closing this support request as completed.

### 1. Connecting pymavlink

QOpenHD is a MAVLink client. There are two different destinations:

- **Read telemetry through OpenHD Ground:** connect to the Ground unit's TCP server on **5760**. For the Ethernet-hotspot address used in the question, use `tcp:192.168.2.1:5760`, not `udpout:192.168.2.1:5760`. Substitute the Ground unit's actual reachable IP.
- **Send a companion's data directly to QOpenHD over an IP network:** send UDP to **the machine running QOpenHD**, port **14550**. That machine may differ from the Ground unit. QOpenHD learns the reply endpoint from received datagrams.

`localhost` always means the machine running the Python script; it does not refer to a remote Ground/QOpenHD device. `0.0.0.0` is a local bind address, not the remote destination. Two programs merely listening with `udpin` will not establish traffic between themselves. Port 14551 is OpenHD Ground's UDP input; port 14550 is its client-output destination / QOpenHD's receive port.

For reading Ground telemetry:

```python
from pymavlink import mavutil

link = mavutil.mavlink_connection(
    "tcp:192.168.2.1:5760", source_system=42, source_component=191
)
heartbeat = link.wait_heartbeat(timeout=10)
if heartbeat is None:
    raise RuntimeError("No heartbeat received; check the Ground IP and connection")
```

For a direct companion-to-QOpenHD example, standard `NAMED_VALUE_FLOAT` messages can carry two development values without creating a new dialect:

```python
import time
from pymavlink import mavutil

# Replace with the reachable IP of the machine running QOpenHD.
link = mavutil.mavlink_connection(
    "udpout:192.168.2.50:14550",
    source_system=42,
    source_component=191,
    dialect="common",
)
started = time.monotonic()
while True:
    link.mav.heartbeat_send(
        mavutil.mavlink.MAV_TYPE_ONBOARD_CONTROLLER,
        mavutil.mavlink.MAV_AUTOPILOT_INVALID,
        0, 0, mavutil.mavlink.MAV_STATE_ACTIVE,
    )
    timestamp_ms = int((time.monotonic() - started) * 1000) & 0xFFFFFFFF
    link.mav.named_value_float_send(timestamp_ms, b"detect_x", 123.0)
    link.mav.named_value_float_send(timestamp_ms, b"detect_y", 456.0)
    time.sleep(1)
```

Use system/component IDs appropriate to your network; do not impersonate the FC or OpenHD systems 100/101. This example requires direct IP reachability. An airborne companion using the OpenHD RF link additionally needs its traffic injected/forwarded through the configured Air/Ground telemetry path; the example does not create that routing.

### 2. Exposing the values to a widget

Adding `L_RO_PROP(int, detect_x, set_detect_x, 0)` only creates a Qt property/setter. It does **not** define a MAVLink message, decode incoming packets, or connect a Python variable to QML.

The implementation path is:

1. Add the properties to an appropriate QObject model. A dedicated companion/detection model keeps this data separate from FC telemetry; `_fcMavlinkSystem` is specifically the flight-controller model.
2. In `MavlinkTelemetry::process_mavlink_message()`, route the intended message and source to that model **before the FC-only system-ID routing**. A separate companion system will not automatically enter the FC handler.
3. Decode `NAMED_VALUE_FLOAT` with `mavlink_msg_named_value_float_decode()`, compare the fixed-length `name` field safely (it is not guaranteed to be null-terminated), and call the appropriate setter. Use floating-point properties if fractional coordinates are needed, and keep model updates on the Qt/model thread.
4. Expose a new model with `engine.rootContext()->setContextProperty(...)` in `app/main.cpp`; bind the widget to that model's properties. Existing `_fcMavlinkSystem` registration shows the pattern.
5. Use `qml/ui/widgets/ExampleWidget.qml` / `BaseWidget.qml` as starting points, register the QML file in `qml/qml.qrc`, and instantiate it in the HUD (`qml/ui/HUDOverlayGrid.qml`).

**The Python example needs the corresponding receiver/model/widget extension above to display the values; stock QOpenHD does not automatically render arbitrary named values.** For a stable structured detection protocol, define a custom MAVLink message and generate matching Python and C/C++ dialect bindings for the sender, QOpenHD, and any intermediate component that parses those messages. QOpenHD uses the `openhd` dialect; `common` is sufficient only for the standard messages in the sending example.

Relevant dev-release sources:
- [Shared ports](https://github.com/OpenHD/QOpenHD/blob/dev-release/app/telemetry/tutil/openhd_defines.hpp)
- [Connection setup and message routing](https://github.com/OpenHD/QOpenHD/blob/dev-release/app/telemetry/MavlinkTelemetry.cpp)
- [UDP transport](https://github.com/OpenHD/QOpenHD/blob/dev-release/app/telemetry/connection/udp_connection.cpp)
- [QML context registration](https://github.com/OpenHD/QOpenHD/blob/dev-release/app/main.cpp)
- [Example widget](https://github.com/OpenHD/QOpenHD/blob/dev-release/qml/ui/widgets/ExampleWidget.qml)
- [Official pymavlink connection/dialect documentation](https://mavlink.io/en/mavgen_python/)

Resolution: connection usage and the custom-widget integration path are documented here. No generic companion detection widget or automatic named-value binding is being claimed as implemented, and this documentation does not claim an end-to-end RF/hardware test.

closed 2026-10-20 by raphaelscholle