OpenHD/QOpenHD #739: Install *.deb, get error

yizhiweimengxiangfendoudefeifei, 2025-07-09:
Hello, when i try "sudo dpkg -i qopenhd_2.6.4-05-27-2025--00-41-53-566dc101_armhf.deb", i get some error message as follows, can you help me?
[image]

raphaelscholle, 2025-07-09:
QopenHD is not made to be just installed ontop of rasbperry pi-os it will cause tons of issues, please just use our images, if you try to do it manually, don't complain that things do not work or work badly. You can add the missing packages from our cloudsmith repo add our repo and execute "apt install openhd"  : curl -1sLf \
  'https://dl.cloudsmith.io/public/openhd/release/setup.deb.sh' \
  | sudo -E bash

yizhiweimengxiangfendoudefeifei, 2025-07-10:
How can i use images?  from this site? 
https://cloudsmith.io/~openhd/repos/dev-release/packages/

raphaelscholle, 2025-07-12:
https://openhdfpv.org/#downloads

closed 2025-07-12 by raphaelscholle